NXP/i3c-slave-design #5: With public release of I3C spec, is the NDA still needed

BurtHarris, 2018-02-26:
Not a big deal, I'm just glad the spec has been made public.

BurtHarris, 2018-08-25:
Hello @pkimelman-nxp, 

It may take a lawyer rather than engineer to figure this out, but what I've been expecting one of several licenses as described under at https://en.wikipedia.org/wiki/BSD_licenses.

The [NXP website documentation](https://www.nxp.com/products/product-information/ip-block-licensing/mipi-i3c:MIPI) indicates that the **Free Basic Slave** IP is under BSD license, but license here is far from a typical BSD agreement.      

Perhaps this indicates the documentation here is out-of-date with NXP's actual licensing policy, or perhaps it's that I've misunderstood the reference to BSD license made on the NXP site.

pkimelman-nxp, 2018-08-25:
MIPI forces NXP to add a “copyright” license on the specification, which creates some restrictions. We need to see if the “open" publication of the I3C spec has a less restrictive license. If so, we will use that. This was not our choice. The current license allows you to use the RTL freely, but limits distribution due to the MIPI constraints. Obviously we preserve the copyright as with Apache/BSD. We did originally have a pure Apache license with typical no-warranty and so on.
When I3C Basic is released, we will then see if we can move back to an Apache license. That should happen in mid to late October.

Regards, Paul

Paul Kimelman – Automotive Platform Architecture
408.906.9676

pkimelman-nxp, 2018-12-14:
The Slave has now been republished with BSD License to go with the release of I3C Basic v1.0. This new publication allow non-members to not only read the spec but build HW from it. So, the RTL is released as BSD again.

pkimelman-nxp, 2019-07-12:
Since the above comment clarified that we now have a normal BSD license, this is closed.

closed 2019-07-12 by pkimelman-nxp

BurtHarris, 2019-07-17:
Thank you @pkimelman-nxp