NXP/i3c-slave-design #7: Basic testbench

matticica, 2019-06-07:
Is there a very basic testbench to simulate the interface?

pkimelman-nxp, 2019-06-07:
At the moment, no. That is available from various companies as well as with the licensed version. We will likely add a simple vector checkout of ENTDAA and some messages. This would show up with the next update - planned for July.

Regards, Paul

Caution: EXT Email

Paul Kimelman – Automotive Platform Architecture
408.906.9676

pkimelman-nxp, 2019-07-12:
The update likely will get moved into Aug. No specific errors to rush it out, and a few new features will show up in the upcoming release and those are being finalized.

rosholm, 2019-08-21:
Does the update include any new features in the I3C slave or is it only new features in the testbench?
Also, is there a roadmap/list of planned new features?

pkimelman-nxp, 2019-08-21:
Yes, the update adds new optional features and the vector test bench.
The main new features are:
1. Slave Reset and SETAASA.
2. More i2c features if needed (e.g. extended address, device ID, SW reset).
3. Some bug fixes.

Plan is early Sept.

Does the update include any new features in the I3C slave or is it only new features in the testbench?

Also, is there a roadmap/list of planned new features?

--
Paul Kimelman – Automotive Platform Architecture
408.906.9676

pkimelman-nxp, 2019-10-31:
The test bench vector set is not yet in the release update, but I will leave the other issue open about this subject. The new update (1.1.11) has features and fixes as noted.

closed 2019-10-31 by pkimelman-nxp