NXP/i3c-slave-design #8: glitch on i3c_data may cause fake clock for clk_SDA

juneysun, 2019-06-24:
We have hooked up the i3c_slave with denali master. For I3C transfer, NXP I3C can get dynamic address assignment from master. However, while I3C master issues write transaction to NXP I3C slave, I3C slave detects wrong STOP bit. It breaks slave’s FSM and slave can not receive data from master. The issue is on the following code:

  always @ (posedge clk_SDA or negedge rst_STOP_n)
    if (!rst_STOP_n)
      STOP_r      <= 1'b0;              // cannot be 1 due to any_START
    else if (pin_SCL_in & ~SDR_hold)    // SDA rise when SCL=high
      STOP_r      <= 1'b1;              // stopped as a "state"

1).  rst_STOP_n is stay as 1
2). Clk_SDA is not toggle, stay as 0
3). SDR_hold = 0 so the ~SDR_hold should have no effect in this case.
4). pin_SCL_in changing 0 -> 1  This caused the STOP_r chaning from 0 -> 1

In our case, the “i3c_sda” line is switching to tri-state. At that moment, denali model also switching from driving a weak1 to strong 0:
“sda” line is also pullup on bench to a pullup strength of 1’b1.
Thus it could be the glitch on SDA line causing the fake clock event.
The SDA can be driven by multiple devices. Is there a way to avoid using SDA as clock? Any suggestion if this is a real issue and how to fix it?

Thanks,

June

pkimelman-nxp, 2019-06-24:
When you use RTL simulations, you have to add some logic to prevent these tiny glitches (e.g. a few ps) because the simulation Flop will trigger. No actual physical flop would, and no pad could react that fast, so this is just an artifact of the RTL sim.
The normal is to make sure the connections are setup so that there is no AND or OR glitch.

Regards, Paul

Caution: EXT Email

Paul Kimelman – Automotive Platform Architecture
408.906.9676

pkimelman-nxp, 2019-07-12:
I am closing it since the explanation was given and no follow up.

closed 2019-07-12 by pkimelman-nxp

Bububhai170, 2024-10-13:
Hi @juneysun juneysun/Paul,
 1. what is the solution for this issue.
 2. where you added delays? can you please share code?
 3. how you used deglitcher filter? can you please share the code to resolve this issue?

waiting for your reply.
Thanks,