NXP/i3c-slave-design #9: I3c slave configured as i2c slave

prats2018, 2019-07-12:
Hi,
I m configuring this nxp slave as  i2c slave with i2c master, Read operation is not happening, I am able to see the data in fb_data, but read is not being initiated. Can you let me know the configuration required to be set for making it work as i2c.

Thanks
Prats

pkimelman-nxp, 2019-07-12:
I am assuming you are using the APB Slave wrapper and not autonomous slave.
If you do not put data into the TX FIFO, it will NACK the read and it will set the ERRWARN bit to tell you that (underrun on address).
The normal model is that they write a single byte (index) and you get that in the RX FIFO. You then push data into the TX FIFO which  is what they read.
The fb_data value should have been pushed into the RX FIFO with fb_data_done. Your app should have seen it then.
Regards, Paul

Caution: EXT Email

Paul Kimelman – Automotive Platform Architecture
408.906.9676

pkimelman-nxp, 2019-07-15:
prats2018, please let me know if this answered the issue or you are still having problems.

closed 2019-08-15 by pkimelman-nxp