NXP/i3c-slave-design #11: ORUN issue

prats2018, 2019-07-19:
Hi,
I am observing tb_orun during read . The description says The internal from-bus buffer/FIFO was overrun (too many chars coming in and not drained by the app fast enough).
How to resolve this.

Thanks
Prathusha

pkimelman-nxp, 2019-07-19:
To bus overrun means you are writing to the TX FIFO too fast. The software should be using the TX status or interrupt. If polling, then read the STATUS register and check of TX has space. If using the interrupt you can fill it the same way. The datactrl register will also tell you how much space.
Note that you can configure the TX FIFO size to be whatever you want.

Caution: EXT Email

prats2018, 2019-07-19:
Thanks, we have increased the Internal FIFO Threshold to 4.So overrun error is resolved.
But, now there is another issue, while reading , we are getting underrun issue..:(
Did we miss any settings? Can you check this waveform attached here?
Note: We are not configuring any APB registers. sdr_Slv_ok  bit is tied to 1. 
[image]

pkimelman-nxp, 2019-07-19:
The underrun simply means it ran out of data, but was not told last one was the last (end). You need to write END if it is the end of the data you want to send. You can either use WDATAB with the END bit, or use WDATABE for this. If using half-words, then WDATAH and WDATAHE.

Caution: EXT Email

[i2c_slave]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41048446%2F61523839-c2839200-aa32-11e9-876c-61eb59d23590.png&data=02%7C01%7Cpaul.kimelman%40nxp.com%7C205d03f774a84d78978608d70c28977d%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C0%7C636991240794113962&sdata=%2BCuvSRws5obssaK3rdRyrLaaHIa0vW5fMZ1MPHOAU8k%3D&reserved=0>

Paul Kimelman – Automotive Platform Architecture
408.906.9676

closed 2019-08-15 by pkimelman-nxp