NXP/i3c-slave-design #13: file missing for fifo support

juneysun, 2019-08-15:
Hi, 

Could you give me some help with the following questions?

1). From the description, we should be able to adjust the FIFO depth. But after I changed the following parameter, I got "Module 'i3c_internal_fb_fifo' and  'i3c_internal_tb_fifo' not defined.". Did I missed some setting during download and where should I get these missing model?
    parameter FIFO_TYPE       = 4'b11_01, // From default 4'b10_00
    parameter  ENA_TOBUS_FIFO = 2,      // From default 0
    parameter  ENA_FROMBUS_FIFO=2,      // From default 0

2). I got "Module 'i3c_time_control' is not defined." if set "parameter ENA_IBI_MR_HJ   = 1, " Is this function not supported in this version or I missed something?

3). I would like to confirm that the ENA_SADDR can only use one selected method.

4). For the 31 parameters defined in the i3c_apb_wrapper, could you help to list which ones maybe updated by user? Or we have to stick with the default value?

Thanks,

June

pkimelman-nxp, 2019-08-15:
Could you give me some help with the following questions?

1). From the description, we should be able to adjust the FIFO depth. But after I changed the following parameter, I got "Module 'i3c_internal_fb_fifo' and 'i3c_internal_tb_fifo' not defined.". Did I missed some setting during download and where should I get these missing model?

You need to include the i3c_internal_fifo.v file in your build. It has the internal FIFO (vs. the ping-pong FIFO/buffer which is local to the data files).

2). I got "Module 'i3c_time_control' is not defined." if set "parameter ENA_IBI_MR_HJ = 1, " Is this function not supported in this version or I missed something?

Time control is not included in the free version, but it is NOT enabled by ENA_IBI_MR_HJ which is allowed. It is enabled by ENA_TIMEC. So, you need to set that to 0.

3). I would like to confirm that the ENA_SADDR can only use one selected method.

Yes. You can select how you want the static address to be defined (or not). If an MMR (or external net if autonomous), then you can change as needed at runtime.

4). For the 31 parameters defined in the i3c_apb_wrapper, could you help to list which ones maybe updated by user? Or we have to stick with the default value?

You may set most of them. The only ones you cannot set are for features not available to the free version. The only things not supported in the free version are: MASTER (ENA_MASTER), HDR-DDR (ENA_HDR), Timing control (ENA_TIMEC), and DMA (SBIF_DMA_b in SEL_BUS_IF).

Thanks,

June

Regards, Paul

juneysun, 2019-08-15:
Hi, Paul:

Thank you very much for the quick response! I agree with the answer 2-4. For the 1st question, these two files are not in the files downloaded a while back. I downloaded the zip again today but still can’t find these two files in the src dir.

Thanks,

June

External Email

pkimelman-nxp, 2019-08-15:
I am unclear why you say that. I just downloaded and unpacked I see the below. You can see the i3c_internal_fifo.v is just below full_wrapper. The two data files (which also use the internal FIFO when selected) are i3c_data_frombus.v and i3c_data_tobus.v which you can see below as well.
[image]

juneysun, 2019-08-15:
Hi, Paul:

Thanks for checking on this. I found these two modules 'i3c_internal_fb_fifo' and 'i3c_internal_tb_fifo' in the i3c_internal_fifo.v. From the documentation, it said “if more than 16 or so are needed, it is recommended to use a separate FIFO scheme”. This is for different memory type instead of flop. It is different with setting the EXT_FIFO, right?

There were two lint Errors that CLK_TST and FIFO_TST out of range in i3c_full_wrapper.v. I modified as 0 to get rid of the lint Error for now. What would be a better way to fix this?
  wire [0:0] bad_clk;
  wire [0:0] bad_fifo;
  assign bad_clk[CLK_TST] = 1'b0;
  assign bad_fifo[FIFO_TST] = 1'b0;

Thanks,

June

External Email

[Image removed by sender. image]<https://user-images.githubusercontent.com/22693343/63123114-28ddd080-bf5d-11e9-8b48-b6f36509ff56.png>

pkimelman-nxp, 2019-08-15:
> Thanks for checking on this. I found these two modules 'i3c_internal_fb_fifo' and 'i3c_internal_tb_fifo' in the i3c_internal_fifo.v. From the documentation, it said “if more than 16 or so are [...]

>> Yes, if you wanted to use a regfile or other method of large FIFO, then use the external FIFO.
> 
> There were two lint Errors that CLK_TST and FIFO_TST out of range in i3c_full_wrapper.v. I modified as 0 to get rid of the lint Error for now. What would be a better way to fix this? 

> These are not lint errors. These are assertions used to catch parameter errors. This is telling you that some config you have is wrong.

> wire [0:0] bad_clk; 
> wire [0:0] bad_fifo; 
> assign bad_clk[CLK_TST] = 1'b0; 
> assign bad_fifo[FIFO_TST] = 1'b0; 
> 

> The CLK_TST says your CLK_SLOW setting is an issue:
  localparam CLK_TST = (|(CLK_SLOW_BITS|CLK_SLOW_MATCH) & ~|ENA_IBI_MR_HJ) ||
                       CLK_SLOW_MATCH>=(1<<CLK_SLOW_BITS);
if CLK_TST evaluates to 1, then there is an issue. So, you do not set CLK_SLOW if you are not using IBI; If you are using IBI, then you need to set the MATCH to be within the BIT size as explained in the micro-arch spec.
The FIFO_TST error is checking for FIFO_TYPE being correct; if you are using EXT_FIFO, then it has to be correct. Check the micro-arch spec for details on how to set these.

juneysun, 2019-08-15:
Hi, Paul:

Thank you for pointing out the *_TST are “assertions used to catch parameter errors”. I will fix them according to the micro-arch spec.

To use external FIFO, there are two separate FIFOs using the following interface signals (PCLK domain), right? We need to have extra logic outside to readout the From-bus FIFO and write to the To-bus FIFO and control the interface signal, right? What does it mean “request model internal” if we set EXT_FIFO=2?
  // Optional external FIFO:
    // To-bus means from slave to master
  input               ixf_tb_avail,     // 1 if byte is available for tobus
  input               ixf_tb_last,      // byte is last for message
  input         [7:0] ixf_tb_data,      // actual data when avail=1
  output               oxf_tb_start,     // pulsed when ixf_tb_avail==1 and START
  output              oxf_tb_used,      // pulsed when tb_data used
    // From-bus means from master into slave
  input               ixf_fb_free,      // 1 if space to take a byte frombus
  output              oxf_fb_req,       // pulse when data to be taken (amd free=1)
  output        [7:0] oxf_fb_data,      // data frombus when req=1
  output              oxf_fb_eof,       // frame ended in repeated START or STOP

Thanks,

June

External Email

pkimelman-nxp, 2019-08-16:
It only affects how the To-bus signal is handled. Between combinatorial and registered. Registered obviously uses an extra cycle in the PCLK domain, but if your path is an issue, it is easier on timing.
  assign avail_tb_ack   = (EXT_FIFO==`EXT_FIFO_REQ) ?
                          (avail_tb_ready & tb_not_full) : // combo
                          ack_push;     // registered

juneysun, 2019-08-16:
Hi, Paul:

Why the FIFO_TYP = 4’b11_01 is considered as Error condition? I think both FromBus and ToBus can both use internal FIFO. The following equation list the FromBus can’t use Internal FIFO.

Localparam = (FIFO_TYPE[2] & ~FIFO_TYPE[0]) ||
                        (FIFO_TYPE[3] & FIFO_TYPE[0]) ||

Thanks,

June

External Email

pkimelman-nxp, 2019-08-16:
Quoting from the micro-arch spec:

•       FIFO_TYPE: this selects which kind of FIFO to use and other FIFO related controls

o   [1:0] = 00 if no FIFO, 01 if internal FIFO, 10 if external FIFO

o   [2] = Add holding buffer to to-bus to avoid direct muxes when internal FIFO used. This is not used when no FIFO or external FIFO – not needed.

o   [3] = Add holding buffer to from-bus when no internal FIFO, which adds 1 more buffer’s worth. If used with external FIFO, separates mux from external FIFO path.

You are not using the bits correctly. This is defined as [3:0], and 4’b11_00 says you have no FIFO at all. The upper two bits are for decoupling.

On Aug 16, 2019, at 2:54 PM, juneysun <notifications@github.com<mailto:notifications@github.com>> wrote:

Caution: EXT Email

Hi, Paul:

Why the FIFO_TYP = 4’b11_01 is considered as Error condition? I think both FromBus and ToBus can both use internal FIFO. The following equation list the FromBus can’t use Internal FIFO.

Localparam = (FIFO_TYPE[2] & ~FIFO_TYPE[0]) ||
(FIFO_TYPE[3] & FIFO_TYPE[0]) ||

Thanks,

pkimelman-nxp, 2019-09-03:
I am assuming the answer was accepted, so closing.

closed 2019-09-03 by pkimelman-nxp