NXP/i3c-slave-design #14: Basic testbench

Thangaselvamd, 2019-08-27:
Is there is any basic testbench for i3c slave design?

Regards,
Thangaselvam D

pkimelman-nxp, 2019-08-27:
We use an extensive test bench and also run commercial VIPs. I cannot release either. For the next release, I am working on a vector test stimulator based on tests we have run.

Sent from my iPhone

On Aug 26, 2019, at 9:20 PM, Thangaselvamd <notifications@github.com<mailto:notifications@github.com>> wrote:

Is there is any basic testbench for i3c slave design?

shanshuixiangyi, 2020-01-09:
I write the VIPs to verify the design,I find some problems about design.

SiliconKite, 2020-07-29:
> I write the VIPs to verify the design,I find some problems about design.

and those problems are .... ?

yangwhtw, 2020-12-08:
Hi, does anyone can provide basic I3C testbench for this slave design? Thanks!