NXP/i3c-slave-design #15: sda_pad_oe

shanshuixiangyi, 2019-09-03:
# Using the  I3C_auton_wrap_full module as the rtl top. 
#  the testcase as follow steps
## 1. Frist,I assign the static address,
## 2. then, I send the static address and Read bit.  
## 3. Dut is ack and enter the ST_READ state.It's okay. 
## 4.  testbeach send scl,The Dut sda_pad_oe control may have a error direction.

The simulation wave as follows.
[image]

shanshuixiangyi, 2019-09-03:
I send the message FF. Sorry

pkimelman-nxp, 2019-09-03:
What is FF?

pkimelman-nxp, 2019-09-03:
> I send the message FF. Sorry

What is FF?

shanshuixiangyi, 2019-09-10:
Read data from bus is FF,So the pad is open drain output. SDA_OE output is low.It's right.

pkimelman-nxp, 2019-09-10:
OK, so I will close it. Thanks.

closed 2019-09-10 by pkimelman-nxp