NXP/i3c-slave-design #18: I3C TIMING CONTROL

shanshuixiangyi, 2019-10-17:
This source do not have

pkimelman-nxp, 2019-10-17:
Correct. The free version does not include  either of the timing control blocks, nor hdr-ddr, nor dma.

Caution: EXT Email

ggpf320, 2019-10-17:
Hi Paul
Can it be understood that the IBI with Mandatory Data Byte is not supported in free version?

pkimelman-nxp, 2019-10-17:
It is supported. The CTRL reg is active to support in the APB version, and the autonomous makes available as well.
Only the time stamp adder is not supported.

Caution: EXT Email

shanshuixiangyi, 2019-10-17:
I have verified the coded.It’s very good!But I’m confused about IBI generation.Why the IBI generation is the clock slow domain?Thank you! 

发自我的iPhone

shanshuixiangyi, 2019-10-17:
Another question about I3C master.I noticed Nxp have a MCU with i3c master.When the MCU go to mass production ？I want to buy this MCU for testing i3c communication.

发自我的iPhone

shanshuixiangyi, 2019-10-18:
> Are you using the autonomous or APB slave? The IBI starts from the system clock so you can initiate it - PCLK or SlowCLK. This is necessary anyway. But it is also because it allows us to pull SDA low even without an SCL - a feature of i3c. That is, when bus is stopped we pull SDA low to get the Master to emit the START clock.
>
>  Sent from my iPhone
>
>
>
>
>  Caution: EXT Email
>
>  I have verified the coded.It’s very good!But I’m confused about IBI generation.Why the IBI generation is the clock slow domain?Thank you!&amp;nbsp;

I use the autonomous as my top module,If my system is only scl clock,we can also use comb logic to pull down.Even we can use asm to generation this start.

发自我的iPhone

pkimelman-nxp, 2019-10-18:
I am not sure what “asm” means here. But, you can generate a fake clock if you want. The CLK_SLOW is there just to synchronize the net.
I agree that for a purely clock-less device which only relies on SCL, this is not generally needed, but that case is hard to see how it can work with IBIs. The normal model was that for clock-less devices such as thermal sensors, they would be polled vs. use an IBI since they have no clock of their own.
CLK_SLOW can be a very slow clock - people use 200KHz and even 32KHz clocks for this.

Regards, Paul

On Oct 17, 2019, at 5:42 PM, Jiahua <notifications@github.com<mailto:notifications@github.com>> wrote:

Caution: EXT Email

I use the autonomous as my top module,If my system is only scl clock,we can also use comb logic to pull down.Even we can use asm to generation this start.

shanshuixiangyi, 2019-10-19:
ASM means asynchronous states machine ,It use comb logic loop to generate.It not use clock 

发自我的iPhone

pkimelman-nxp, 2019-10-19:
ASM means asynchronous states machine ,It use comb logic loop to generate.It not use clock&nbsp;

OK. The problem is that we need a synchronizer normally, as a flop goes metastable if you happen to change its D input during the setup period.
In your case, you would need to have the synchronizer used for IBI work with your async mechanism.
You will see this wherever we cross between SCL and CLK_SLOW.

--
Paul Kimelman – Automotive Platform Architecture
408.906.9676

shanshuixiangyi, 2019-10-20:
Okay,Thanks.

发自我的iPhone

closed 2019-10-31 by pkimelman-nxp