NXP/i3c-slave-design #19: Are unhandled CCCs passed through to the mr_* bus?

kurtrad1, 2019-10-23:
I do not see a way to configure this I3C so that it passes any unimplemented CCCs through to the magic bus.  Am I missing something?  I would like to implement a global CCC in the vendor reserved space.

pkimelman-nxp, 2019-10-23:
I do not see a way to configure this I3C so that it passes any unimplemented CCCs through to the magic bus. Am I missing something? I would like to implement a global CCC in the vendor reserved space.

If you enable for CCCs, it always does.
    parameter MAGIC_RULES     = 4'b0000,// bits for rules. [0]=RW vs. WO, [1]=CCCs, [2]=no idx
Set the CCC one. It will pass all unhandled CCCs through. The mr_trans_ccc_done port is pulsed to indicate it is a CCC.

kurtrad1, 2019-10-23:
Thanks for the reply.  I was looking in the wrong place which was the ENA_CCC_HANDLING parameter.

closed 2019-10-31 by pkimelman-nxp