NXP/i3c-slave-design #20: SDA negated prematurely during ACK

bbsbucket, 2019-10-25:
The SDA negation by slave for ACK in response to 7'h7E seems to be premature due to the following in i3c_sdr_slave_engine.v.  SDA is negated when SCL related to ACK is asserted by the master which causes STOP_r and the state machines returns to state 0.  This is for PIN_MODEL=PINM_COMBO.

assign SDA_oe = ...
                  ((state==ST_ACK_NACK)  & ack_ok &       
                   (~in_i3c_msg | (~event_drive & (~pin_SCL_in | was_read)))) |
                           …

Any insight will be appreciated.

pkimelman-nxp, 2019-10-25:
>> The SDA negation by slave for ACK in response to 7'h7E seems to be premature due to the following in i3c_sdr_slave_engine.v. SDA is negated when SCL related to ACK is asserted by the master which causes STOP_r and the state machines returns to state 0. This is for PIN_MODEL=PINM_COMBO.

>> assign SDA_oe = …
>> ((state==ST_ACK_NACK) & ack_ok &
>> (~in_i3c_msg | (~event_drive & (~pin_SCL_in | was_read)))) |
>> …

>> Any insight will be appreciated.

That is not correct. The I3C Master is required to do the handoff on the SCL High period. The model of I3C, unlike i2c, is that he Master matches the ACK state on/after the rising edge of SCL. Both Master and Slave will hold it for some period, but the Slave is obligated to release it by the minimum 1/2 SCL period minus setup (so at most by less than all of the SCL High period, as seen by the Master); so, the Master will hold it Low if the Slave had ACKed (made Low), or the Master will hold High (by Pullup or driver) if the Slave had NACKed (did not drive, so pullup made High).
So, the Slave must release SDA on SCL rising (as seen by the Slave, so SDA High-Z will be probably 12ns or more after the master drove SCL high).
I hope that clarifies. You will see an explanation of this in the I3C spec.
Regards, Paul

bbsbucket, 2019-10-25:
Thank you for your prompt reply explaining the ACK negation protocol.  I have verified it (Sec 5.1.2.3.1) which clearly states what you have explained above.
Ref:
1. The I3C Slave shall hold the SDA line Low during the ACK (while SCL is Low). 
· This is an Open Drain SCL Low period. 
2. After the I3C Slave sees the rising edge of SCL, it releases the SDA line to High-Z. 
· The I3C Slave shall release the SDA line using normal (Push-Pull) timing (release the SDA line 
 as soon as it sees SCL rising). 
3. After the rising edge of SCL, the I3C Master shall drive the SDA line Low.
· As a result, both Master and Slave will be driving the SDA line Low for a short overlap (which 
 is safe).

closed 2019-10-31 by pkimelman-nxp