NXP/i3c-slave-design #21: Request to have source uploaded as separate files instead of a zip file

ilesser, 2019-10-29:
Hi, 

First of all thanks for sharing this design with the community, as time goes by more and more people embrace the open hardware!

Would it be possible to have the code uploaded as separate files instead of having one zip file?

This distribution approach makes diffing new changes more complicated and I think it goes against the whole point of having a version control system.

Thanks,
Ignacio

pkimelman-nxp, 2019-10-29:
I have been looking at how to easily transfer over the files and directories. The zip is created by automation from a different source control system. There is a directory which is zipped, so it would be possible, but it does not look like I can just "drop" a directory into Github. 
I assumed people would just pull the new zip and diff the files from the previous pull to see what has changed and/or merge in any changes they have made.

pkimelman-nxp, 2019-10-29:
I should clarify one point: every file will look different to GitHub every time because the copyright header will reflect the tag change.

ilesser, 2019-10-30:
Well it turns out a directory can be dropped into github, just run `git add <directory>` and then commit.
If someone wants to use this repository as a submodule inside another git project it requires an extra step of unzipping the file which is something very uncommon.
The most common way of handling submodules (and frankly any git repository) is to just have the source available.

Anyway, this is just a request/suggestion that would make things easier to interface with other projects.

Thanks again!

pkimelman-nxp, 2019-10-31:
Well it turns out a directory can be dropped into github, just run git add <directory> and then commit.
If someone wants to use this repository as a submodule inside another git project it requires an extra step of unzipping the file which is something very uncommon.
The most common way of handling submodules (and frankly any git repository) is to just have the source available.

Anyway, this is just a request/suggestion that would make things easier to interface with other projects.

As you perhaps can tell, we do not use Git for HW designs, only for SW. So, the process has been to extract from the HW source control, and then assemble into a zip and ‘drop' into GitHub using the browser interface. There is no local git command available in the env.
It looks like this should work to drop a folder, but it was not working before with Safari; I will try again to see if this works now with the new OS and all.
If so, I would put it under a directory so that people can choose. From what I can tell, a number of people who are getting this zip are not Git users either - this is just an easy way to transfer it.
Regards

pkimelman-nxp, 2019-10-31:
I have uploaded the unzipped version of the files into “unzipped”. Let me know how this works for you. I will always update both for now since the zip file can be compared quickly to be sure the original.
Regards, Paul

closed 2019-10-31 by pkimelman-nxp

ilesser, 2019-11-01:
Thanks! 
That's exactly what I needed.

Regards,
Ignacio