NXP/i3c-slave-design #23: Unsupported syntax in Verilog 2005

rosholm, 2019-11-12:
The code contains 'standalone' generate blocks (no conditions) which are only allowed in Verilog 2001 but not in Verilog 2005.

These blocks are located here:
i3c_sdr_slave_engine.v: line 780
i3c_daa_slave.v: line 314
i3c_ccc_slave.v: line 539 and 599

All are related to the free version cut.
The issues were found when trying to synthesize the design using Synopsys DC compiler.

BR
/Anders

kurtrad1, 2019-11-12:
We are working on getting the autonomous slave to synthesize in Incentia designcraft.  We are currently debugging this.  What tool are you using to synthesize this at NXP?

pkimelman-nxp, 2019-11-12:
We use both DC and RC across the company. Most of the uses of I3C are with DC.

>> We are working on getting the autonomous slave to synthesize in Incentia designcraft. We are currently debugging this. What tool are you using to synthesize this at NXP?

pkimelman-nxp, 2019-11-12:
> The code contains 'standalone' generate blocks (no conditions) which are only allowed in Verilog 2001 but not in Verilog 2005.

> These blocks are located here:
> i3c_sdr_slave_engine.v: line 780
> i3c_daa_slave.v: line 314
> i3c_ccc_slave.v: line 539 and 599

> All are related to the free version cut.
> The issues were found when trying to synthesize the design using Synopsys DC compiler.

I have never seen an issue. We use Verilog 2001 and System Verilog (you can just tell it to use SV even though .v vs. .sv).
I have not even heard of 2005 to be honest. Seem strange that they would disallow generate without a condition or loop.
This is from code-removal for the free version (some features are removed). I will look at moving the free version cut to outside the generate/endgenerate.
For now, you can use verilog 2001 or SV.

Regards, paul

pkimelman-nxp, 2019-12-12:
The cutlines are now positioned so an if() is present in the generate, even though empty.

closed 2019-12-12 by pkimelman-nxp