NXP/i3c-slave-design #24: Latches inferred on dma_r[5:4]

rosholm, 2019-12-11:
Line 703 to 718 in i3c_regs.v infer latches on dma_r[5:4].

(We use SEL_BUS_IF = 5'b01111 and USE_D_RESET is not defined)

It looks like a tool issue since the latches are gone when adding
```
	  else
	    dma_r[5:4] <= dma_r[5:4];
```
after line 711.

See Spyglass and Design Compiler reports below.

Spyglass (SpyGlass_vN-2017.12-SP2) report:

```
############### Non-BuiltIn -> Goal=lint/lint_rtl ###############
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
ID       Rule                 Alias                 Severity    File         Line    Wt    Message
======================================================================================
[1B4]    InferLatch           (OpenMORE 5.5.2.1)    Error       /i3c_regs.v  705     5     Latch inferred for signal '\dma_reg.dma_r [5:4]' in module 'i3c_regs'
```

Synopsys Design Compiler (2018.06-SP3) report:

```
Inferred memory devices in process
        in routine i3c_regs_1_000000000000_000_00000000_00_2_55_00_2_000000_1f00_0_0_6_2f_0_10_0000000_8_8_0_0_0_0f_d_0_2_2_0_02_0000_0 line 703 in file
                'i3c_regs.v'.
===============================================================================
|    Register Name    |   Type    | Width | Bus | MB | AR | AS | SR | SS | ST |
===============================================================================
|  dma_reg.dma_r_reg  |   Latch   |   1   |  N  | N  | Y  | N  | -  | -  | -  |
|  dma_reg.dma_r_reg  |   Latch   |   1   |  N  | N  | N  | Y  | -  | -  | -  |
|  dma_reg.dma_r_reg  | Flip-flop |   4   |  Y  | N  | Y  | N  | N  | N  | N  |
===============================================================================
```

pkimelman-nxp, 2019-12-11:
This is a clear tool bug that seems to have appeared in a recent update. The DC does the right thing since it eventually collapses the [5:4] flops to constant 0s. I believe someone has filed a report with Synopsis, but I would encourage doing so as well. This is clearly wrong. The logic is a clock block, so cannot be viewed as a latch, which is not clock edge sensitive. That the assignment of non-0 is controlled by a constant param (and so may well be 0 and so the logic goes away) should not be viewed as changing anything. The synthesis tool should remove all conditional logic which it knows will never be true. So, the end result is that the flops never change from 0 and so become constants in the netlist.

 

Caution: EXT Email

pkimelman-nxp, 2019-12-12:
As noted, this a tool bug and should be reported to your EDA vendor. This is fully valid RTL (Verilog) and there is no excuse for their claiming there are latches. The use of a constant in an if() in a clock block simply should remove the conditional logic, but does not change the flops.

closed 2019-12-12 by pkimelman-nxp