NXP/i3c-slave-design #26: Push pull and open drain

minijose, 2019-12-24:
Hi,

Could you please tell me how push pull and open drain is managed in I3C slave rtl verilog code?
Please bear with me if its a silly question. As I am new to I3c didn't get much idea about push pull and open drain. 

Thanks and Regards,
Mini Jose

pkimelman-nxp, 2019-12-24:
Could you please tell me how push pull and open drain is managed in I3C slave rtl verilog code?
Please bear with me if its a silly question. As I am new to I3c didn't get much idea about push pull and open drain.

The Slave block connects to your SDA pad using an output-enable (_oena) and an output (_out) net. Output-enable switches between High-Z and output driven using 0 and 1 (if your pad has oena_n, just invert the net before it arrives at the pad). So, Open-drain sets Output to 0 (Low) and flipping output-enable toggles between High-Z and Low as usual. Push-pull simply leaves output-enable as 1 (driven) and flips output between 1 (Hgh) and 0 (Low). 

The input from the SDA should be connected to the pad Schmitt triggered input. 

Thanks and Regards,
Mini Jose

minijose, 2020-01-08:
Thank you for the detailed explanation..
So in Open-drain, sda_oena  will be high, whenever it wants to drive the sda ouput to 0. Else sda_oena will be low. 
 When I checked the I3C slave rtl code, sda output is assigned to 1 when  sda_oena is low.
As you explained, when sda_oena is 0, sda output should have driven high-z instead of 1.
Please correct me if am wrong.

pkimelman-nxp, 2020-01-09:
Thank you for the detailed explanation..
So in Open-drain, sda_oena will be high, whenever it wants to drive the sda ouput to 0. Else sda_oena will be low.
When I checked the I3C slave rtl code, sda output is assigned to 1 when sda_oena is low.
As you explained, when sda_oena is 0, sda output should have driven high-z instead of 1.
Please correct me if am wrong.

You are correct in what you are saying but you seem to be making incorrect assumption. OENA means output-enabled. If OENA=0, then the state of OUT is irrelevant since the output is disabled (it is in High-Z/Input). So, OUT has no meaning in  that case. When OENA=1, OUT will be used, and in this case, OUT will be 0.