NXP/i3c-slave-design #27: i2c_spike_ok error

shanshuixiangyi, 2020-01-09:
When I do fpga verification, 
First,I use I2c to communication,It's works well. 
Second,I enter i3c mode,it's fine.  
Third, I send RSTDAA CCC command,then,I use I2c communication . It's fail.  
Because I2c_spike_ok signal is error.Thanks.

pkimelman-nxp, 2020-01-09:
When I do fpga verification,
First,I use I2c to communication,It's works well.
Second,I enter i3c mode,it's fine.
Third, I send RSTDAA CCC command,then,I use I2c communication . It's fail.
Because I2c_spike_ok signal is error.Thanks.

The rule is that i2c_spike_ok is on when the Slave starts in case you are on an I2C legacy bus. As soon as it sees START,7E, it turns it off and never turns it back on. This is correct because you know you are on an I3C bus, even if in I2C mode (from RSTDAA) so the spike filter is NOT appropriate. The I3C slave is expected to be responsive to many I3C CCCs in push-pull mode even if operating off its i2c address.

So, the behavior is correct.

shanshuixiangyi, 2020-01-09:
When I do RSTDAA command，I change the mode into I2c mode.The device should response the I2C write or read.If the i2c_spike_ok never turn off(filter never turn on), The I2c communication will be fail.I change the logic.If RSTDAA into I2c,I will turn off the i2c_spike_ok signal.Then see 7E,I turn on I2c_spike_ok again.That's fpga works fine.I think the I3c device should have the ability of work on the I2c mode if we want.

pkimelman-nxp, 2020-01-10:
When I do RSTDAA command，I change the mode into I2c mode.The device should response the I2C write or read.If the i2c_spike_ok never turn off(filter never turn on), The I2c communication will be fail.I change the logic.If RSTDAA into I2c,I will turn off the i2c_spike_ok signal.Then see 7E,I turn on I2c_spike_ok again.That's fpga works fine.I think the I3c device should have the ability of work on the I2c mode if we want.

That is absolutely wrong. The spike filter has nothing to do with a successful or unsuccessful i2c message. The spike filter adds a 50ns plus latency delay to smooth out noise spikes. An I3C bus should not have noise spikes. So, whether a 50ns filter is on or off, the i2c slave will work correctly. 

So, you have some other issue in your test if you think the spike filter is needed for i2c to work. 

RSTDAA causes the I3C Dynamic address to be erased and so the I3C slave will simply listen for the original i2c static address (if any). The spike filter has no bearing on its ability to do so. 

I suggest you determine how you made it dependent on a spike filter – presumably a bug in your test setup. For example, if you are assuming that the Slave cannot respond sooner than 50ns after the SCL falling edge, that would be a mistake in your assumption – i2c does not have any such rule.

shanshuixiangyi, 2020-01-10:
The  spike filter is designed for I2C mode,When you go back to I2C mode,Using I2C mode to communication without spike filter may have some issues. 
I don't think have some problems with the i2c_spike_ok logic，I think using I2c mode with spike filter is right.I have not seen any description with i2c_spike_ok(RSTDAA i2c_spike_ok must be turn off) in the MIPI I3C spec.Thank you.Next month i will tape out.

pkimelman-nxp, 2020-01-10:
The spike filter is designed for I2C mode,When you go back to I2C mode,Using I2C mode to communication without spike filter may have some issues.

No!!! That is not valid. The spike filter only exists to mitigate noise in a long wire i2c bus. It has NO other purpose. I3C buses are shorter and are assumed to be routed to be less exposed to noise. So, you do not need the filter. Many people use i2c today without filters by the way. For this case, we allow you to use it on a legacy i2c bus. Once we know we are on an I3C bus, we do not need it.

I don't think have some problems with the i2c_spike_ok logic，I think using I2c mode with spike filter is right.I have not seen any description with i2c_spike_ok(RSTDAA i2c_spike_ok must be turn off) in the MIPI I3C spec.Thank you.Next month i will tape out.

I3C spec only requires legacy i2c device to have a spike filter, and that is only so they do not see the high speed push-pull traffic.

shanshuixiangyi, 2020-01-10:
Thank you for your patience in answering my questions. Have you ever mass produced this code into chips?

pkimelman-nxp, 2020-01-10:
Thank you for your patience in answering my questions. Have you ever mass produced this code into chips?

Yes. It is in wide production by multiple companies, including NXP.