NXP/i3c-slave-design #28: Wrong parameter default MAP_DA_AUTO

sdolezal, 2020-02-03:
https://github.com/NXP/i3c-slave-design/blob/cd9582ecb345bad1a5fa34af11d8e2443ccc23f2/unzipped/src/i3c_full_wrapper.v#L106

Like mentioned in the "Integration Guide", bits 12:8 of MAP_DA_AUTO should be set to '1' if Auto Map DAA is not used. Wouldn't it be more consistent if those bits were set to '1' by default since DAA enable bit 2 is '0' by default? I got compilation errors by using the current defaults.
This affects apb, full and slave wrapper.

pkimelman-nxp, 2020-02-03:
Yes, I have not updated GitHub, but the correction used is in the apb_wrapper:
    parameter  MAP_DA_AUTO    = {5'd1,1'b0,1'b0,3'd0,1'b0,1'b0,1'b0},
This ensures the PID_CNT is valid all the way down.

closed 2020-02-03 by pkimelman-nxp

sdolezal, 2020-02-04:
I'm using the full wrapper, not the APB wrapper. Also, if someone uses the slave wrapper directly, the same problem occurs. The best solution would be to fix this in every hierarchy level, not just in top. Thanks