NXP/i3c-slave-design #31: The function of HJ? thank you very much

owen-yang0726, 2021-01-11:
Hi dear,
  Does free version of i3c_slave_design provide the function of HJ?

I try to modify the parameter of ENA_IBI_MR_HJ to 5'b01000, ENA_TIMEC to 6'd0(due to we do not have design of time control).

My test flow is set 1'b1 to address of 0x4 for bit[0] SLVENA, then trigger my master VIP write data to slave by I2C protocol with static address. 

Next, I try to set 2'b11 to address of 0xc for bit[1:0] EVENT after my master VIP write data finish and pull stop.

But I found the SDR_hold will pull 1'b1 when I set 2'b11 to address of 0xc for bit[1:0] EVENT, and I can't no see slave design try to   issue start.

In my point, I think slave determines that the Bus is idle it may request a start by pulling SDA low after I set 2'b11 to address of 0xc for bit[1:0] EVENT and then issue address of 7'h02 to request a DAA process. In the end turn to master VIP to choose ack/nack the slave request?

Can you please tell me what's wrong with my setting flow? Does anything I misunderstanding? or the free version does not support HJ? Attachment as my waveform shown that time 17164ns issue STOP and Time 22760ns et 2'b11 to address of 0xc for bit[1:0] EVENT, and then SDR_hold pull up, and then never seen slave design try to issue stop...

Thank you very much. many thanks.

[image]

pkimelman-nxp, 2021-01-11:
Just start the Hot-Join before you slave-enable. You cannot Hot-join after as it thinks it is already on the bus.

owen-yang0726, 2021-01-12:
Hi Paul,
  Thank for your response. Actually I modify my I3C master VIP and test flow.  It seem does work, i3c slave design can trigger start and emit 7'h02 hot join request to master. Can you please help me check my test flow is correct or not?
1. I set bit[31:25] for static address and set 1'b1 for bit[0] for SLVENA to same address of 0x4(CONFIG)
2. Trigger my I3C master VIP to transmit data to I3C slave design by I2C protocol with static address
3. I3C master VIP write data finish and trigger stop (SDA&SCL keep high)@time 17211ns
4. Compare RX fifo data to data of I3C master VIP transmission by address of 0x40 RDATAB bit[7:0] DATA0
5. I set 2'b11(hot-join) to address of 0xc for bit[1:0] EVENT after compare rx fifo data
6. I3C slave trigger start and issue 7'h02 HJ request after 1us @time 92148ns
7. I3C VIP master ack request and trigger stop
8. I3C VIP master trigeer start and send ENTDAA CCC command....

I found design signal of hold_engine will pull high and slv_enable wii pull down as soon as I set 2'b11 to address of 0xc for bit[1:0] EVENT.
Though the signal of cf_slvena will keep high if I start slave_enable before start hot-join.
But HJ enable result in hold_engine become high and slv_enable become low,
Therefore start slave-enable before hj start seem also does work?

Thank you very much. many many thanks.
Best regards,
Owen