NXP/i3c-slave-design #33: "i3c_time_control" module is missing

huangluyang0, 2022-05-18:
Dear IP owner, when I tried to use i3c_auton_wrapper with "ENA_TIMEC" parameter enabled, which means that I wanted to enable the Timing control/Timestamping feature, it turned out that "i3c_time_control" module is missing, I think that module is necessary for i3c slave timing control, like sc1/sc2/timec_ibi_byte. I'll really appreciate it if you can also include the "i3c_time_control" module in the package, thanks!!!

tsuhuai, 2022-05-27:
I have implemented a sample i3c_time_control.v. Can I upload it here?

pkimelman-nxp, 2022-05-29:
Please create a contrib directory for that. It needs to be clear it is sourced from elsewhere and that we have not checked it.

tsuhuai, 2022-05-30:
OK.
By the way, I have problems verifying static address function for autonomous slave.
Can you help me?

Following are the tasks I issued.  And I get a wrong ACK at 2nd ADDR_HEADER task

task START
task ADDR_HEADER(7'h7e, WRITE);
task CHK_ACK
task `CCC_SETAASA
task START_r
task ADDR_HEADER(7'h7e, WRITE);
task CHK_ACK
task WRITE_8BITS
task CHK_ACK
task STOP

tsuhuai, 2022-05-30:
Also, please check the file 'i3c_auto_wrap_full.v'
There are logic or "|" operator at both the end of line 142 and beginning of line 143.

My simulator: cadence IUS9 .

tsuhuai, 2022-06-11:
Should I pull a request when I finished it?