NXP/i3c-slave-design #34: GETPID CCC not executed

sdolezal, 2022-05-24:
https://github.com/NXP/i3c-slave-design/blob/e82a728f93f9a7d82d2948c7d84f217dec4d4343/unzipped/src/i3c_sdr_slave_engine.v#L829

Hi Paul,

please check this line again. I doubt that it's correct when compared to the meaning of cccmask register. Could it be, that this should be ccc_uh_r instead?

I continue to see problems when sending GETPID to the target device. The IP decides not to execute GETPID if CCCMASK is a register and bit 0 is set. I expect it to execute it nonetheless, since CCCMASK only should affect CCCs which are not executed directly by the IP (which is NOT the case here).

Please see also my former issue on this: https://github.com/NXP/i3c-slave-design/issues/30

It seems, that GETPID works sporadically depending on the data which is sent after 0x8D (in my case the target address). So maybe this tricked me in believing, that the problem was solved.

Best regards,
Stephan

pkimelman-nxp, 2022-05-24:
Yes, I agree it is incorrect. It shoudl be ccc_uh_r but I also need to check as I think that needed some tweaking. I will be creating a new update of this distribution next month, which will roll up some bug fixes and a few changes around v1.1.1 of the spec.

huangluyang0, 2025-05-21:
@pkimelman-nxp I saw the version was not updated since 2020, were you planning to make a new release as mentioned here on 25th May 2022?