NXP/i3c-slave-design #36: Autonomous slave, IBI, no MDB, ACK pulse handoff

tsuhuai, 2022-06-06:
Hi:
Have you ever test the IBI without MDB condition?
ENA_IBI_MR_HI = 6'b00_0001;//IBI capable
ENA_TIMEC =5'b0_0000;//no Timing control
BCR[2]=0;//no MDB

My  task pulls the SDA low 20ns before the (ACK)  SCL 0->1 transition, and then keep it High-Z for 20ns when SCL is high.
However, the slave did handoff.

If I set BCR[2]=1, ENA_IBI_MR_HJ=6'b00_0011 and perform the simulation again, correct ACK handoff take places.
DId I miss anything?

pkimelman-nxp, 2022-06-06:
Is has been tested in the past, but no user is using it without MDB, so it is possible something got broken. 
It should release/handoff after the SCL rising edge of the Controller ACK, as you say.

tsuhuai, 2022-06-07:
Would you confirm that ? Thanks.

tsuhuai, 2022-06-07:
Also, in my simulation , I found there is no ACK handoff for Hot Joint event. 
Since Slave engine didn't pull SDA low. Is it correct?