NXP/i3c-slave-design #38: the difference between PIN_MODEL == `PINM_COMBO and PINM_REG

wh0xjjx, 2022-12-23:
Hi Paul,
     could you explain the different between PIN_MODEL == `PINM_COMBO and PINM_REG,  i am a little confused, thank you!

pkimelman-nxp, 2022-12-27:
Pin combo generates the pad Controls live. Ends up with smaller logic since can be spread over path. But need to be careful in timing closure. Works well with older process but newer/smaller process is too variant often.
Registered means the controls are registered a half clock in advance.

Caution: EXT Email

wh0xjjx, 2022-12-30:
Thank you for your reply.
I test combo and reg . the pin_SDA_out and pin_SDA_oena are almost same.
but when state == R9TH, the  pin_SDA_oena seems fall down too early(in the posedge of SCL)
and the parity value on pin_SDA_out should be high, while it is low, a little wired

wh0xjjx, 2022-12-30:
The result is in here!
[image]
[image]